OpenAstroTech/OpenAstroTracker-Firmware #6: "AccelStepper" can be improved

jeromefavrou, 2020-12-13:
the "AccelStepper" library can be improved. This one accumulates in time a software error. It can be interesting to mathematically integrate this one. Where the "AccelStepper" lib makes 

```
    unsigned long time = micros();   
    if (time - _lastStepTime >= _stepInterval)
    {
	if (_direction == DIRECTION_CW)
	{
	    _currentPos += 1;
	}
	else
	{
	    _currentPos -= 1;
	}
	step(_currentPos);

	_lastStepTime = time; 
	return true;
    }
```

I had coded a version with this fix, my lib is much less complete than AccelStepper but takes into account the software error. As a comparison in my lib I did like this . "this->m_ei" is integrate error

```
unsigned long r_elsp= micros() - this->m_elaps;

        if( (_mv_type == MOVE_TYPE::HALF_STEP ? this->m_inter_delay/2.0 : this->m_inter_delay)  <=r_elsp-this->m_ie)
        {
          this->next_step(_mv_type);
          this->m_elaps = micros();
          this->m_ie += this->m_inter_delay-r_elsp;

          step++;
        }
```

andre-stefanov, 2020-12-18:
We have analyzed the code and you are right, there is definitely an error which stacks with each step because the `_lastStepTime` is based on the current time instead on the interval and the `runSpeed()` function is not called on each microsecond.

jeromefavrou, 2020-12-27:
I made a pull on the lib accelstepper's git
https://github.com/jeromefavrou/AccelStepper/pull/1

andre-stefanov, 2021-12-21:
We are moving away from AccelStepper dependency meanwhile because it is not performant enough and has several issues. Also the linked PR was not merged by that library meaning it is not supported enough.

closed 2021-12-21 by andre-stefanov