OpenAstroTech/OpenAstroTracker-Firmware #75: ESP32 with "default" DISPLAY_TYPE_LCD_KEYPAD

Viper-Gtr, 2021-03-11:
Hi,
so i just finnished my mechanical build of this project. Now its time to jump into the firmware.
Is there a reason, why i can't compile the firmware for a ESP32 with the default "LCD Keypad Shield"?
Normamly these two should work, or is there something i missed?

ClutchplateDude, 2021-03-11:
Why can't you? What's stopping you?

ClutchplateDude, 2021-03-11:
We don't support that config because it requires too many pins in the already limited ESP32 pin availability. But if you solder it to the pins you want and define the pins, you can use it. The Shield is meant for the Mega though, since you just plug it onto the Mega, no soldering required.

Viper-Gtr, 2021-03-14:
Done like you said. Works fine.
I'm developing a "shield" with the naked ESP/WROOM 32, so it has enough pins available. But i'm struggeling eith the UART part of the TMC drivers.

closed 2021-03-21 by ClutchplateDude