OpenAstroTech/OpenAstroTracker-Firmware #88: Devel branch seems to be building all versions

rwillett, 2021-04-18:
I'm trying to build the OpenAstroTech firmware for a Makerbase V2.1 board with TMC2209 stepper drivers. I am unclear if these are UART drivers or not, however thats a different problem.

I downloaded the OAT dev branch and unpacked it to a folder.

I then went to https://config.openastrotech.com/ and went through each step to create the Configuration_local.hpp file. I have guessed that the TMC2209 are UART.

I created the file 'Configuration_local.hpp' and pasted in the output from the https://config.openastrotech.com/ tool. This looks OK on the surface but what do I know .... It has the right board definitions in and the rest look sensible.

I renamed Configuration_sample_local.hpp to OLD_Configuration_sample_local.XXX to avoid any conflicts.

I click on the blue tick button the bottom left of VSC and a build process starts. Indeed it appears that a build process starts in turn for every motherboard 
```
mega2560       SUCCESS   00:00:12.378
mksgenlv21     SUCCESS   00:00:11.255
mksgenlv2      SUCCESS   00:00:11.009
mksgenlv1      SUCCESS   00:00:11.110
esp32          SUCCESS   00:00:36.410
```

I'm sure there is something I have done wrong but unclear what mistkae I have made.

Any help welcomed.

Thanks

Rob

Here's the Configuration_local.hpp if that helps....
```
/////////////////////////////////////////////////////////////////////////////////////////////////////////
// This configuration file was generated by the OAT Configurator at https://config.openastrotech.com for
// firmware V1.9.00 and later.
// Save this as Configuration_local.hpp in the folder where you placed the firmware code.

// We live in the Northern Hemisphere
#define NORTHERN_HEMISPHERE 1

// We are using the MKS GEN L V2.1 board
#define BOARD BOARD_AVR_MKS_GEN_L_V21 

////////////////////////////////
// RA Stepper configuration 
 // See supported stepper values. Change according to the steppers you are using
// Using the NEMA 17, 0.9°/step stepper for RA
#define RA_STEPPER_TYPE STEPPER_TYPE_NEMA17

// Using the TMC2209-UART driver for RA stepper motor
#define RA_DRIVER_TYPE DRIVER_TYPE_TMC2209_UART

// Define some RA stepper motor settings
#define RA_MOTOR_CURRENT_RATING      900 // mA
#define RA_OPERATING_CURRENT_SETTING 80 // %
#define RA_STEPPER_ACCELERATION 3000
#define RA_STEPPER_SPEED 1200
#define RA_SLEW_MICROSTEPPING 8
#define RA_TRACKING_MICROSTEPPING 64
// #define RA_SERIAL_PORT Serial3  // You may need to uncomment or change this, depending on how you wired the UART
 // TMC2209 Stealth Mode (spreadCycle) - More precise tracking when not in stealth mode (set to 0), but steppers will make a high-pitched sound.
#define RA_UART_STEALTH_MODE   1

// Using the 16 tooth purchased gear for RA belt
#define RA_PULLEY_TEETH 16

////////////////////////////////
// DEC Stepper configuration 
 // See supported stepper values. Change according to the steppers you are using
// Using the NEMA 14, 0.9°/step stepper for DEC
#define DEC_STEPPER_TYPE STEPPER_TYPE_NEMA17

// Using the TMC2209-UART driver for DEC stepper
#define DEC_DRIVER_TYPE DRIVER_TYPE_TMC2209_UART

// Define some DEC stepper motor settings
#define DEC_MOTOR_CURRENT_RATING      900 // mA
#define DEC_OPERATING_CURRENT_SETTING 80 // %
#define DEC_STEPPER_ACCELERATION 3000
#define DEC_STEPPER_SPEED 1200
#define DEC_SLEW_MICROSTEPPING 16
#define DEC_GUIDE_MICROSTEPPING 64
// #define DEC_SERIAL_PORT Serial3  // You may need to uncomment or change this, depending on how you wired the UART
 // TMC2209 Stealth Mode (spreadCycle) - More precise tracking when not in stealth mode (set to 0), but steppers will make a high-pitched sound.
#define DEC_UART_STEALTH_MODE   1

// Using the 16 tooth purchased gear for DEC belt
#define DEC_PULLEY_TEETH 16

////////////////////////////////
// Display configuration 
// Define the type of display we are using. Currently: LCD Shield w/ keypad
#define DISPLAY_TYPE DISPLAY_TYPE_LCD_KEYPAD

////////////////////////////////
// GPS Addon configuration 
// Define whether we have the GPS addon or not. Currently: No GPS
#define USE_GPS 0

////////////////////////////////
// Digital Level Addon configuration 
// Define whether we have the Digital Level or not. Currently: No Digital Level
#define USE_GYRO_LEVEL 0

////////////////////////////////
// AutoPA Addon configuration 
// Define whether we have the AutoPA add on or not. Currently: No AutoPA
#define USE_AZIMUTH_ALTITUDE_MOTORS 0
```

rwillett, 2021-04-18:
I've just looked at platform.ini and have added 
```
default_envs = mksgenlv21 
```

and now it only builds my version of the code. Suspect that is the answer?

cameronswinoga, 2021-04-18:
Yes by default `platformio` builds all environments, this is the expected behavior. You can specify a specific environment on the command line with:
```
pio run -e mega2560
```

rwillett, 2021-04-18:
Thanks. That makes sense. I've updated platform.ini to just tell it to build for the MKS 2.1 board. 

Now if I could only get the Mac version of VSC to recognise the device, I'd be happy. Am trying on a Windows box to see if that works.

ClutchplateDude, 2021-04-18:
For questions like this (support, essentially) our Discord server would be the better place....

rwillett, 2021-04-18:
Ok, thanks. I did try to use Discord and it seemed to want an invitation and then it didn;t work and then I gave up on it....

is there an easy way to get into Discord that doesn't require brain surgery?

Thanks

Rob

ClutchplateDude, 2021-04-18:
I don't think so, a hundred+ people have made it in so far without problems....

[Later: I see you made it]

rwillett, 2021-04-18:
Sorry, I had an awful experience with Discord for another project and I still fail to see why people clearly like it. Hey ho, I like Emacs and for some reason, some people hate that editor :)

Rob

closed 2021-04-18 by rwillett