OpenAstroTech/OpenAstroTracker-Firmware #103: Upload fail =(

nafital, 2021-05-24:
Environment    Status    Duration
-------------  --------  ------------
mega2560       SUCCESS   00:00:29.547
mksgenlv21     SUCCESS   00:00:29.410
mksgenlv2      SUCCESS   00:00:29.538
mksgenlv1      SUCCESS   00:00:29.738
esp32          FAILED    00:00:35.691
====================================================================== 1 failed, 4 succeeded in 00:02:33.924 ======================================================================The terminal process "C:\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

Help ****

// We live in the southern hemisphere

#define NORTHERN_HEMISPHERE 0

 

// We are using an ATMega 2560 card (or clone)

#define BOARD BOARD_AVR_MEGA2560

 

//////////////////////////////////////////

// RA Stepper configuration

// See the allowed stepper values. Change according to the steppers you are using

// Using NEMA 17, 1.8 ° / step for RA

# define RA_STEPPER_TYPE STEPPER_TYPE_NEMA17

#define RA_STEPPER_SPR 200

 

// Using the TMC2209-UART driver for RA stepper motor

#define RA_DRIVER_TYPE DRIVER_TYPE_TMC2209_UART

 

// Using the purchased 16-tooth gear for the RA belt

#define RA_PULLEY_TEETH 16

nafital, 2021-05-24:
mega2560 SUCCESS 00: 00: 29.547
mksgenlv21 SUCCESS 00: 00: 29.410
mksgenlv2 SUCCESS 00: 00: 29.538
mksgenlv1 SUCCESS 00: 00: 29.738
esp32 FAILED 00: 00: 35.691
================= ================================= ========================= ===== 1 failed, 4 succeeded at 00: 02: 33.924 ======== ============================== ==================== ====================== The terminal process "C: .platformio \ penv \ Scripts \ platformio .exe 'run', '--target', ' upload '"ended with exit code: 1.

Terminal will be reused for tasks, press any key to close it.

cameronswinoga, 2021-05-24:
Can you attach your `Configuration_local.hpp` please?

andre-stefanov, 2021-05-24:
Wait, why is it building for all the platforms? looks like you did not set the desired board in platformio. The error is probably because it builds for all platforms including esp32, which does not support everything from atmega2560

nafital, 2021-05-31:
//////////////////////////////////////////////////////// //////////////////////////////////////////////////////// ///// 

// Este arquivo de configuração foi gerado pelo Configurador OAT em https://config.openastrotech.com para 

// firmware V1.9.00 a V1.9.06. 

// Salve como Configuration_local.hpp na pasta onde você colocou o código do firmware. 

 

// Vivemos no hemisfério sul 

#define NORTHERN_HEMISPHERE 0 

 

// Estamos usando a placa ATMega 2560 (ou clone) 

#define BOARD BOARD_AVR_MEGA2560 

 

////////////////////////////////////// 

// Configuração do RA Stepper  

// Veja os valores de stepper suportados. Mude de acordo com os steppers que você está usando 

// Usando o NEMA 17, 1.8 ° / stepper para RA 

# define RA_STEPPER_TYPE STEPPER_TYPE_NEMA17 

#define RA_STEPPER_SPR 200 

 

// Usando o driver TMC2209-UART para motor de passo RA 

#define RA_DRIVER_TYPE DRIVER_TYPE_TMC2209_UART 

 

// Definir algumas configurações do motor de passo RA 

# define RA_MOTOR_CURRENT_RATING 900 // mA 

# define RA_OPERATING_CURRENT_SETTING 80 //% 

#define RA_STEPPER_ACCELERATION 3000 

# define RA_STEPPER_SPEED 1200 

#define RA_SLEW_MICROSTEPPING 8 

# define RA_TRACKING_MICROSTEPPING 64 

// TMC2209 Modo Stealth (spreadCycle) - Quando definido como 0, o rastreamento é mais preciso, mas com ruído (som agudo). Quando definido como 1, eles são silenciosos. 

#define RA_UART_STEALTH_MODE 0 

 

// Usando a engrenagem comprada de 16 dentes para a correia RA 

#define RA_PULLEY_TEETH 16 

 

////////////////////////////////////// 

// Configuração DEC Stepper  

// Veja os valores de stepper suportados. Mude de acordo com os steppers que você está usando 

// Usando o NEMA 17, 1.8 ° / stepper para DEC 

# define DEC_STEPPER_TYPE STEPPER_TYPE_NEMA17 

#define DEC_STEPPER_SPR 200 

 

// Usando o driver TMC2209-UART para DEC stepper 

#define DEC_DRIVER_TYPE DRIVER_TYPE_TMC2209_UART 

 

// Definir algumas configurações de motor de passo DEC 

#define DEC_MOTOR_CURRENT_RATING 900 // mA 

#define DEC_OPERATING_CURRENT_SETTING 80 //% 

#define DEC_STEPPER_ACCELERATION 3000 

#define DEC_STEPPER_SPEED 1200 

#define DEC_SLEW_MICROSTEPPING 16 

#define DEC_GUIDE_MICROSTEPPING 64 

// TMC2209 Modo Stealth (spreadCycle) - Quando definido como 0, o rastreamento é mais preciso, mas com ruído (som agudo). Quando definido como 1, eles são silenciosos. 

#define DEC_UART_STEALTH_MODE 0 

 

// Usando a engrenagem comprada de 16 dentes para a correia DEC 

#define DEC_PULLEY_TEETH 16 

 

////////////////////////////////////// 

// Configuração de exibição  

// Defina o tipo de exibição que estamos usando. Atualmente: LCD Shield com teclado 

#define DISPLAY_TYPE DISPLAY_TYPE_LCD_KEYPAD 

 

////////////////////////////////////// 

// Configuração do complemento GPS  

// Defina se temos ou não o complemento GPS. Atualmente: Sem GPS 

#define USE_GPS 0 

 

////////////////////////////////////// 

// Configuração do complemento de nível digital  

// Defina se temos o Nível Digital ou não. Atualmente: Sem Nível Digital 

#define USE_GYRO_LEVEL 0 

 

////////////////////////////////////// 

// Configuração do complemento AutoPA  

// Defina se temos o complemento AutoPA ou não. Atualmente: Sem AutoPA 

#define AZIMUTH_ALTITUDE_MOTORS 0 

 

#define DEBUG_LEVEL (DEBUG_NONE)

nafital, 2021-05-31:
> Você pode anexar seu `Configuration_local.hpp`favor?

//////////////////////////////////////////////////////// //////////////////////////////////////////////////////// ///// 

// Este arquivo de configuração foi gerado pelo Configurador OAT em https://config.openastrotech.com para 

// firmware V1.9.00 a V1.9.06. 

// Salve como Configuration_local.hpp na pasta onde você colocou o código do firmware. 

 

// Vivemos no hemisfério sul 

#define NORTHERN_HEMISPHERE 0 

 

// Estamos usando a placa ATMega 2560 (ou clone) 

#define BOARD BOARD_AVR_MEGA2560 

 

////////////////////////////////////// 

// Configuração do RA Stepper  

// Veja os valores de stepper suportados. Mude de acordo com os steppers que você está usando 

// Usando o NEMA 17, 1.8 ° / stepper para RA 

# define RA_STEPPER_TYPE STEPPER_TYPE_NEMA17 

#define RA_STEPPER_SPR 200 

 

// Usando o driver TMC2209-UART para motor de passo RA 

#define RA_DRIVER_TYPE DRIVER_TYPE_TMC2209_UART 

 

// Definir algumas configurações do motor de passo RA 

# define RA_MOTOR_CURRENT_RATING 900 // mA 

# define RA_OPERATING_CURRENT_SETTING 80 //% 

#define RA_STEPPER_ACCELERATION 3000 

# define RA_STEPPER_SPEED 1200 

#define RA_SLEW_MICROSTEPPING 8 

# define RA_TRACKING_MICROSTEPPING 64 

// TMC2209 Modo Stealth (spreadCycle) - Quando definido como 0, o rastreamento é mais preciso, mas com ruído (som agudo). Quando definido como 1, eles são silenciosos. 

#define RA_UART_STEALTH_MODE 0 

 

// Usando a engrenagem comprada de 16 dentes para a correia RA 

#define RA_PULLEY_TEETH 16 

 

////////////////////////////////////// 

// Configuração DEC Stepper  

// Veja os valores de stepper suportados. Mude de acordo com os steppers que você está usando 

// Usando o NEMA 17, 1.8 ° / stepper para DEC 

# define DEC_STEPPER_TYPE STEPPER_TYPE_NEMA17 

#define DEC_STEPPER_SPR 200 

 

// Usando o driver TMC2209-UART para DEC stepper 

#define DEC_DRIVER_TYPE DRIVER_TYPE_TMC2209_UART 

 

// Definir algumas configurações de motor de passo DEC 

#define DEC_MOTOR_CURRENT_RATING 900 // mA 

#define DEC_OPERATING_CURRENT_SETTING 80 //% 

#define DEC_STEPPER_ACCELERATION 3000 

#define DEC_STEPPER_SPEED 1200 

#define DEC_SLEW_MICROSTEPPING 16 

#define DEC_GUIDE_MICROSTEPPING 64 

// TMC2209 Modo Stealth (spreadCycle) - Quando definido como 0, o rastreamento é mais preciso, mas com ruído (som agudo). Quando definido como 1, eles são silenciosos. 

#define DEC_UART_STEALTH_MODE 0 

 

// Usando a engrenagem comprada de 16 dentes para a correia DEC 

#define DEC_PULLEY_TEETH 16 

 

////////////////////////////////////// 

// Configuração de exibição  

// Defina o tipo de exibição que estamos usando. Atualmente: LCD Shield com teclado 

#define DISPLAY_TYPE DISPLAY_TYPE_LCD_KEYPAD 

 

////////////////////////////////////// 

// Configuração do complemento GPS  

// Defina se temos ou não o complemento GPS. Atualmente: Sem GPS 

#define USE_GPS 0 

 

////////////////////////////////////// 

// Configuração do complemento de nível digital  

// Defina se temos o Nível Digital ou não. Atualmente: Sem Nível Digital 

#define USE_GYRO_LEVEL 0 

 

////////////////////////////////////// 

// Configuração do complemento AutoPA  

// Defina se temos o complemento AutoPA ou não. Atualmente: Sem AutoPA 

#define AZIMUTH_ALTITUDE_MOTORS 0 

 

#define DEBUG_LEVEL (DEBUG_NONE)

nafital, 2021-05-31:
> Espere, por que está construindo para todas as plataformas? parece que você não configurou a placa desejada na plataforma. O erro é provavelmente porque ele é compilado para todas as plataformas, incluindo esp32, que não oferece suporte a tudo, desde atmega2560

Project Tasks> env:mega2560> General> Build    =(

cameronswinoga, 2021-05-31:
@nafital

1. If possible, please use English for comments, it looks like you have some translation software that is messing up replies and your code
2. When adding code to a comment, please use three backticks (` ``` `) before and after so that it shows up as a "code block" and not normal comment text
3. Can you open up a command prompt, change to the `OpenAstroTracker-Firmware` directory, and run the command: `pio run -e mega2560 -t upload`

nafital, 2021-05-31:
@cameronswinoga 

I managed to solve this problem =) obg.
but now my Nema17 RA engine is screaming =S

cameronswinoga, 2021-05-31:
Yes, the noise that the stepper motors are making are because `UART_STEALTH_MODE` is disabled. This makes the motors more powerful (easier to maintain position accuracy) but a bit louder.

closed 2021-05-31 by cameronswinoga

nafital, 2021-05-31:
@cameronswinoga that strange in this most current version I can't connect with the stellarium

ClutchplateDude, 2021-06-01:
Support is available on Discord. This is not sn issue. Closing.

nafital, 2021-06-01:
@ClutchplateDude 
good morning can you pass me this discord? THX

nafital, 2021-06-03:
@cameronswinoga good night can you pass me the DISCORD so I can solve this problem? connection via stellarium thx

ClutchplateDude, 2021-06-03:
There are links on the wiki and on reddit. On phone sorry.