OpenAstroTech/OpenAstroTracker-Firmware #107: Crash if attempting to use an alternate display configuration, 20 x 4

g2maurer, 2021-06-03:
My design uses an esp32 with a 20x4 I2C LCD. When I added startup code for the new display type in b_setup.hpp (~ line 24) I used the display's size, 20 columns and 4 rows.   LcdMenu lcdMenu(20, 4, MAXMENUITEMS);
This causes the firmware to crash because in the file LcdMenu.cpp the function LcdMenu::updateDisplay() allocates a char buffer (bufMenu[17]) with a size that is based on a column size of 16 (all hard coded). This buffer should instead be sized based on the LCD configuration  provided in LcdMenu::lcdMenu(col, row, maxmenuitems) 
OR 
better yet the display size should be defined in Configuration_local.hpp.
I am attempting to correct this and at the same time I am separating the display and input (buttons) configurations so that any combination of display and input devices can be accommodated.

ClutchplateDude, 2021-06-03:
You probably want to follow the example of the other display type. All the menu code is optimized for 16x2. You want to create a new LCD class altogether that handles the interface.

g2maurer, 2021-07-06:
Decided not to pursue.

closed 2021-07-06 by g2maurer