OpenAstroTech/OpenAstroTracker-Firmware #121: Inverted GoTo. South America

nafital, 2021-07-13:
https://user-images.githubusercontent.com/84690582/125535131-4463fcb2-91af-4b21-95da-fc2265aed324.mp4

I don't know why I live in Brazil. =(

trishmapow, 2021-07-14:
Hi. I can't quite see the firmware information there. Are you fully updated? And to confirm, you are in the Southern Hemisphere part of Brazil? I believe this issue should've been fixed, but I personally haven't tried with Stellarium. Please also attach the OATControl logs here (click on the log files button).

nafital, 2021-07-14:
> Oi. Não consigo ver as informações do firmware lá. Você está totalmente atualizado? E para confirmar, você está na parte do hemisfério sul do Brasil? Acredito que esse problema deveria ter sido corrigido, mas pessoalmente não tentei com o Stellarium. Anexe também os logs do OATControl aqui (clique no botão arquivos de log).

good night I'm only Brazil.
I'm using the latest firmware update 1.9.16, but this problem is happening on any system, in ota control, stellarium or skytech x

nafital, 2021-07-14:
OTAControl log 
[OATControl_2021-07-13_19-13-24-Paulo.log](https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/files/6812901/OATControl_2021-07-13_19-13-24-Paulo.log)

trishmapow, 2021-07-14:
Hi, please download the latest develop code here: <https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/archive/refs/heads/develop.zip> The issue should be fixed.

nafital, 2021-07-15:
> Baixe o código de desenvolvimento mais recente aqui: https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/archive/refs/heads/develop.zip O problema deve ser corrigido.

I installed the firmware, it worked more or less but I had to leave the DEC inverted, press it up, it goes down, but it's hitting the goto position now thx

trishmapow, 2021-07-18:
I just opened a PR for that issue #123. You can  manually edit the file if you want, for now.

trishmapow, 2021-08-18:
@nafital Do you still notice any bugs with the latest firmware? If not, you can close the issue.

lfpaca, 2021-10-11:
Hello all, I am also in the south hemisphere and I can confirm it now works. Last official release is is indeed reversed, but the development branch fixes the issue.

lfpaca, 2022-06-17:
I was also having erratic behaviors from the OAT in the south hemisphere (firmware V1.11.5) and had to carry some changes to the code. I did not dig into the algorithm enough to be able to affirm that what I did did not create any other problems. What was happening is that the mount was solving all targets as it were in the northern hemisphere so I had to do two things: flip the DEC by 180 degrees and mirror the RA. All the changes were done in the Mount.cpp module. For the RA I had to change the conditional in line 1197 (flipped > to <):

if (NORTHERN_HEMISPHERE ? degreePos < 0 : degreePos < 0)
    {
        hourPos += 12;
        if (hourPos > 24)
            hourPos -= 24;
    }

I also got rid of the conditional in line 3279 which was adding 12h to the RA value. These pieces of code were flipping the RA position. For the declination I had to add a new conditional related to the NOTHERN_HEMISPHERE right after line 3316, adding 180 degrees to the variable "moveDEC".

I did some testing and it appears to be working fine now, but as I said I am not familiar enough with the code to guarantee I did not create other problems.

BeccaraNZ, 2022-08-05:
I can confirm I was having this issue, using the above info I did the following:

Code changes in Mount.CPP:

~line 1200, "if (NORTHERN_HEMISPHERE ? degreePos < 0 : degreePos > 0)" changed to "if (NORTHERN_HEMISPHERE ? degreePos < 0 : degreePos < 0)"
~line 3280 commented out "moveRA += 12;" with a //
~line 3300 added "moveDEC += 180" immediately under "float moveDEC = decTarget.getTotalDegrees();"

I also ended up having to flag my RA as inverted, so far this has been working. I will change the RA stepper wiring as soon as I can to confirm if the RA inversion code is needed or not

BeccaraNZ, 2022-08-07:
User GaryMCK on Discord was having issues with SYNC causing a dec flip, I confirmed the issue and applied a fix that works for me. I've compiled all my mount.cpp changes into a diff. Please excuse if it's in the wrong format as this is my first attempt at contributing code here.

~line 1260 
// Fix for southern hemisphere issues
    if (!NORTHERN_HEMISPHERE)
    {
    decAdjust += 180;
    }
    

[agBEBCZa.txt](https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/files/9276141/agBEBCZa.txt)