OpenAstroTech/OpenAstroTracker-Firmware #132: Problems on the DEC axis for the southern hemisphere

allafs, 2021-08-18:
Hi. My name is Armando Leonardo.
Sorry for the English, but I'm using Google Translate to help me.
I live in the southern hemisphere, in Brazil, and I have a problem moving the DEC axis. As expected and described on the website, the DEC axis appears on the LCD -90º. However, when I move it to a target, only the DOWN button obeys, negatively incrementing the value, that is, the OAT only increments values from -91 to -270, when it should implement values from -89 to 0 and 360 to 270.
I've already checked the behavior of the buttons, as per the troubleshooting guidelines, but everything is OK with the buttons.
It seems to me that there is a problem in the programming, which does not include the southern hemisphere, because of the use of the (-) sign.
Would you be able to help me?
Thanks.

https://user-images.githubusercontent.com/43935909/129978168-7f2842cc-11da-4c14-b16e-d7ea4b3424d1.mp4

trishmapow, 2021-08-18:
Hi, what firmware version are you using? Please try with the latest develop code and let me know if it is still an issue: https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/archive/refs/heads/develop.zip.

allafs, 2021-08-19:
Hi. I'm using version 1.9.19.
I'll try as you suggested and let you know.
Thanks.

allafs, 2021-08-19:
Trishmapow, I did as you directed me.
I installed the new version, 1.9.28, and even installed the new version of IDE Arduino.
Unfortunately it still has the same problem.
I would like to add that in the CTRL, MANUAL SLEWING menu, both the DOWN and UP buttons work normally.
The problem is only the inclusion of the DEC coordinates to move the axis to the target.