OpenAstroTech/OpenAstroTracker-Firmware #148: Develop / Arduino 1.9.34 does not start on MKS Gen_L V2.1

michael-ring, 2021-12-01:
I have built both develop and release version, created Configuration_local.hpp with Weblink and followed 

https://wiki.openastrotech.com/Boards/MKS_Gen_L

to wire up the display but the display stays empty. 

I have also tweaked debugging to 
#define DEBUG_LEVEL (DEBUG_ANY)

but I do not see any feedback on serial as well (set to 19200)

Is there a pre-compiled firmware somewhere that I can try?

To build I used Arduino IDE and set BOARD:

#ifndef BOARD
  #define BOARD BOARD_AVR_MKS_GEN_L_V21
#endif

andre-stefanov, 2021-12-01:
Did you check the contrast of your display? there should be a potentiometer (little screw on a blue box) which you can turn to change contrast of the display. It can take many turns until the display starts to display something.

michael-ring, 2021-12-01:
Ok, now I officially feel stupid, after playing with the potentiometer I can actually see that the firmware runs. Thank you for your help!

closed 2021-12-01 by michael-ring