OpenAstroTech/OpenAstroTracker-Firmware #154: RAMPS won't build

crashmaxx, 2022-01-14:
Trying to build V1.10.2

I use the the online firmware configurator and get the attached Configuration_local.hpp
[Configuration_local_hpp.txt](https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/files/7872975/Configuration_local_hpp.txt)

When I try to build using PIO it fails for all environments.

If I use the Configuration_sample_local.hpp instead, then the mega2560 environment successfully builds.

If I change the line
`#define BOARD BOARD_AVR_MEGA2560`
to
`#define BOARD BOARD_RAMPS`
it again fails for all environments.

This is the first error I get, with a lot of similar ones after
`In file included from src\../LocalConfiguration.hpp:20:0,
                 from src\../Configuration.hpp:50,
                 from src\Declination.cpp:1:
src\../Configuration_local.hpp:24:0: error: "BOARD" redefined [-Werror]
 #define BOARD BOARD_RAMPS`

I noticed Constants.hpp does not have a #define BOARD_RAMPS line or anything similar.

CTetford, 2022-01-19:
There was some naming inconsistencies between the config generator and the firmware. 

Try downloading the latest FW again and starting with a fresh config from https://config.openastrotech.com/

crashmaxx, 2022-01-20:
Went through the config generator again and I am now able to compile V1.10.3 without errors.

Thanks for fixing this!

closed 2022-01-20 by crashmaxx