OpenAstroTech/OpenAstroTracker-Firmware #186: Problems creating a working config for mega2500 L293 Motorshield

Thrillerninja, 2022-06-02:
Hi, is there any guide how to create a config by yourself, or even a finished config for using a [L293](https://elektro.turanis.de/html/prj144/index.html) Motor Shield? I have big problems figuring out where to change stuff needed to get it working

julianneswinoga, 2022-06-02:
Since the L293 uses a shift register to interface with the microcontroller it is incompatible with our stepper library (which needs direct access to step and direction pins). Sorry :frowning_face:

Thrillerninja, 2022-06-03:
How difficult do you think it would be to change the Stepper lib?

julianneswinoga, 2022-06-03:
You can check it out yourself, _most_ of the stepper code is in `src/Mount.cpp` and it's not very well abstracted. It would be a lot of work just to support an obscure piece of HW for one person :grin:

Thrillerninja, 2022-06-03:
Yeah, but going to look into it. Thanks for the help