OpenAstroTech/OpenAstroTracker-Firmware #190: RA Stepper Motor with Sideral Tracking

AstrAuDobson, 2022-08-09:
I have a problem with the RA motor in sidereal tracking. Motor does not run smoothly. I can hear (but it's very slight) little clicks (about 10-15), then the motor stops, then it runs steadily (1-2 seconds) and it starts again like this. I am with arduino MEGA, A4988, NEMA17 1.8°. I tried different stepper motor cards, different motors, to play on the value of the microstepping, nothing changes.

beebrain, 2022-08-10:
Did you check the A4988 current?  In my opinion, I think the current is not enough. Try to adjust the Vref in A4988. ref https://ardufocus.com/howto/a4988-motor-current-tuning/

AstrAuDobson, 2022-08-15:
I have check with 12v power supply and set Vref = 0.8v, it's really better.

I'll buy 0.9deg stepper motors and TMC2209

CTetford, 2022-10-10:
Closing as it looks like this is resolved.

closed 2022-10-10 by CTetford