OpenAstroTech/OpenAstroTracker-Firmware #191: Compiling Error redefinition of 'float fabsf'

sniperlucian, 2022-08-13:
trying to compile with Arduino IDE 1.8.19 on Linux (Manjaro).

Get following compile error:

```

OpenAstroTech/OpenAstroTracker-Firmware/src/Utility.cpp:212:7: error: redefinition of 'float fabsf(float)'
  212 | float fabsf(float x)
      |       ^~~~~
In file included from OpenAstroTech/OpenAstroTracker-Firmware/src/Utility.cpp:1:
/usr/avr/include/math.h:163:21: note: 'float fabsf(float)' previously defined here
  163 | static inline float fabsf (float __x)
      |                     ^~~~~
exit status 1
Error compiling for board Arduino Mega or Mega 2560.
```

julianneswinoga, 2022-08-13:
Hmm, can't replicate on my end. Arduino IDE 1.8.19 on Manjaro as well:
https://gist.github.com/julianneswinoga/b2ca9b2b51629eee0e0d570969521e47

Could you try platformio? It's really easy on linux:
```bash
$ cd OpenAstroTracker-Firmware
$ python3 -m venv .venv
$ ./.venv/bin/pip install platformio
$ ./.venv/bin/pio run -e ramps
```

julianneswinoga, 2022-10-03:
Closing because no response

closed 2022-10-03 by julianneswinoga

sniperlucian, 2022-10-03:
sorry - completely missed the replay. 

compilation run through with platformio.

julianneswinoga, 2022-10-03:
> Closing because no response

Sorry, you're saying that it worked with platformio?

sniperlucian, 2022-10-04:
yes it worked with platformio. 

- made a fresh git clone 
- generated basic config file with web interface (doesn't work with default config - missing parameter) 
- followed your steps 
- compilation run successful trough 

guess I need to take a deeper look at platformio - its new to me ;)