OpenAstroTech/OpenAstroTracker-Firmware #192: XSDLL & XSDLU - ignoring argument

mwmason, 2022-09-16:
[Issue 158](https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/pull/158) added the ability to set DEC lower and upper limits based on a parameter.  However Mount::setDecLimitPositionAbs always sets the limits based on the constants DEC_LIMIT_UP and DEC_LIMIT_DOWN, ignoring the passed argument stepperPos:

`_decUpperLimit = DEC_LIMIT_UP * _stepsPerDECDegree;`

and:

`_decLowerLimit = -(DEC_LIMIT_DOWN * _stepsPerDECDegree);`

Should this be adjusted to allow for either the limit and/or the passed argument?

anjok, 2022-09-17:
When you is this, you also need to change the value when you get it back for display and when you retrieve it

anjok, 2022-09-19:
This here seems to work for me: 

https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/compare/develop...anjok:OpenAstroTracker-Firmware:indi-fix

It won't interfere with current configs and will now give back the degree value instead of steps. However, it's really unclear what should be in degrees and what is not. So some renaming, doc changes etc would also be in order.

Likewise I doubt that setting the current pos is correct, as it's given as "stepperPos" and then again multiplied by stepsPerDegree

CTetford, 2022-10-13:
Fixed in #198.

closed 2022-10-13 by CTetford