OpenAstroTech/OpenAstroTracker-Firmware #199: System freeze after Firmware  "update"

stephanp66, 2022-10-12:
Hello all, 
I have re-flashed the mksV2.1 board  FW (1.11.5) because i swapped the Axes of the Digital level.
Yesterday system works fine.

After this wenn turn on OAT the Display und system works, Powerled is on.
Moving RA/Dec axis to home, that  works fine.
Digital leveling works fine.
because Indoor GPS is not found, press 2x the right button to go on.
then after this the display shows RA DEc and 01h etc 

A few sec later the system freeze , no Displaybutton respond, and the display shows no h.
On the MKS Board the blue led next to powerled and red led next to mosfet turn on
HE1 goes from  0V to 12V

I used the firmware configurator and flashed it several times
result is always the same.

OpenAstrotech Discord :
https://discord.com/channels/693401540744183859/1027676652718792714

-----
Each Time i made a "clean all" bevor build and upload.
I tested several combinations of add-ons.
All Add-ons alone work without problems.
If there are two, there are no problems, no matter what combination.

Only if I use all three (gps, leveling, AutoPA) together
then the system break after skipping the "GPS-Sat search" (indoors) with the R Button when it reaches the "main menu".

julianneswinoga, 2022-11-04:
Sorry, was this resolved @stephanp66 ? There hasn't been any major FW changes since you had this problem?

stephanp66, 2022-11-04:
Sorry, was this resolved ?
Yes and No. :)
I used the OAT after this without the Digital Leveling, Ohnly GPS and AutoPA. And it worked fine.
After a Week or so my Dec Motor got hot and noisy (it was a Cheap from China) so i changed it to a Quality Motor.
Then i thougt to try again with all Addons and activated the Digital leveling in the Configuration.local
build and uploadet it and it worked without errors.