OpenAstroTech/OpenAstroTracker-Firmware #210: RA homing off the rails

anjok, 2022-12-11:
When I set the speed/accel of the new stepper lib too high (I used 16.0 for both), MHRR3 doesn't find the magnet anymore and subsequently goes off the rails on the other side (which is actually a second issue as it shouldn't do so).

Issuing :RM# before the homing makes it kinda work again, but it doesn't move to 0 afterwards. Apparently the RA moves are "weird". When I give MXr100, it moves 100 steps. When I give MXr200, it just doesn't stop until I issue Q.

ClutchplateDude, 2022-12-11:
What version of the FW?

anjok, 2022-12-11:
The one with the new stepper lib. develop branch

ClutchplateDude, 2022-12-11:
I didn’t ask what branch, I asked what version

ClutchplateDude, 2024-05-12:
I mean if it is moving to fast to detect the magnet, there isn't much we can do :-), since that's a hardware issue, then. MKS might be too slow to detect both transitions.