OpenAstroTech/OpenAstroTracker-Firmware #215: Feature Request - Hall Sensor Homing Error depending on direction.

changedsoul, 2023-01-01:
IM not sure what this translates into error looking at the sky, but linearly on the OAT, my sensors seems to have some hysteresis depending on the direction being homed.
If I home in one direction, and then home in the opposite direction, my home positioning will be different by about a 1/8 - 1/4 inch.

Im wondering if its possible to do a double home....find sensor and move past, then move back the other way and move past, and fine the middle of the two directions.

I believe currently it just find the start, looks for the end, and then moved to the middle of that. Problem is, that position changes depending on what direction you begin the homing sequence. 
Perhaps its the extremely cheap hall sensors I have.

ClutchplateDude, 2023-01-01:
This could very well be, but the solution you suggest makes it even more inaccurate. Best results are going to be gotten by always going from the same side.

changedsoul, 2023-01-02:
It would be better if it came from the same direction indeed, but it may not always come from the same direction.
For example, with people who are using dual DEC steppers that balance out the ring weight, it may not always be sagging low on the left side. 
If the ring is rotated more east (magnet already past the sensor), the ring will rotate east first, then reverse direction to find the sensor. It then finds center based on the reverse direction, instead of always finding center by moving east.

As a hack and test, I modified the HallSensorHoming.cpp line 218 from
                        _homingData.state                               = HomingState::HOMING_FINDING_END;
to
                        _homingData.state                               = HomingState::HOMING_MOVE_OFF;

It seems to force it to look for the sensor and then act like its starting to home if it was already over the sensor. This seems to force it move west past the sensor and then begin the normal home in the east direction.

Im not a great programmer and dont understand a lot of what is going on, Im just tinkering, but does this seem like a decent change to force it to always home in the same direction no matter where its starting from?

closed 2023-01-02 by changedsoul