OpenAstroTech/OpenAstroTracker-Firmware #220: Focus Pin Configuration for MEGA2560 double with RA and DEC pins

AnsgarSchmidt, 2023-01-15:
Hi,

When I have a look at the board configuration for the MEGA2560 board I see double pin assignments and I wonder which one are correct:
[pins_MEGA2560.hpp](https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/blob/develop/boards/AVR_MEGA2560/pins_MEGA2560.hpp)
```
FOCUS_STEP_PIN  32  is the same pin as RA_MS1_PIN   32
FOCUS_DIR_PIN     33  is the same pin as DEC_MS1_PIN 33
FOCUS_EN_PIN      34  is the same pin as RA_MS2_PIN    34
```
Thanks
Ansgar

AnsgarSchmidt, 2023-01-15:
May I suggest PIN 47, 48, 49 they are free and very close on the ATMEGA board to the other pins (far backside pin connector)

CTetford, 2023-01-15:
The pins_MEGA2560.hpp pins file has been deprecated since V1.10.

When using an Arduino Mega a RAMPS should be used as well. It uses pins_RAMPS.hpp instead.

The previous method of using separate driver boards was prone to wiring/hardware issues so it was changed. The pins file is still there for archive purposes but really should be removed at this point.

closed 2023-01-15 by CTetford

AnsgarSchmidt, 2023-01-15:
Thanks for your very fast answer. When building my own hardware which board pin should I use / is the most stable / tested one? 
**AVR_MKS_GEN_L_V21**  or **RAMS**?

Thanks a lot
Ansgar

CTetford, 2023-01-15:
Either is supported, it depends on the hardware you're using. 

If you need any help with assembly/electronics, the Discord server is the best place to ask questions.