OpenAstroTech/OpenAstroTracker-Firmware #225: esp32 build broken with version 1.12.6

dc6jn, 2023-04-10:
Hi,
it seems to me that with changes made from v 1.12.5 (d5daf2e1) to 1.12.6beta (38586cea) my esp32 build crashes with a

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400dde4a  PS      : 0x00060e30  A0      : 0x800d9308  A1      : 0x3ffb2140
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000844  A5      : 0x3ffb2110
A6      : 0x00000025  A7      : 0x3ffc2900  A8      : 0x800dea0c  A9      : 0x3ffb2130
A10     : 0x3ffb217c  A11     : 0x0000002b  A12     : 0x0beb37e9  A13     : 0x3ffc2900
A14     : 0x00000025  A15     : 0x3ffb9303  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000020  LBEG    : 0x40086004  LEND    : 0x4008600f  LCOUNT  : 0x00000000

Backtrace: 0x400dde47:0x3ffb2140 0x400d9305:0x3ffb2170 0x400daeaf:0x3ffb21b0 0x400db372:0x3ffb21f0 0x400d2012:0x3ffb2220 0x400e030a:0x3ffb2290

  #0  0x400dde47:0x3ffb2140 in AccelStepper::stop() at .pio/libdeps/esp32/AccelStepper/src/AccelStepper.cpp:666
  #1  0x400d9305:0x3ffb2170 in Mount::stopSlewing(int) at src/Mount.cpp:2329 (discriminator 2)
  #2  0x400daeaf:0x3ffb21b0 in Mount::configureHemisphere(bool, bool) at src/Mount.cpp:227 (discriminator 2)
  #3  0x400db372:0x3ffb21f0 in Mount::configureRAStepper(unsigned char, unsigned char, unsigned int, unsigned int) at src/Mount.cpp:269 (discriminator 2)
  #4  0x400d2012:0x3ffb2220 in setup() at src/b_setup.hpp:312 (discriminator 3)
  #5  0x400e030a:0x3ffb2290 in loopTask(void*) at .platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

My guess is that with the changes made to the stepper driver things got broken.
Thanks!

dc6jn, 2023-04-10:
found a possible cause (and solution):
configureHemisphere(inNorthernHemisphere, true);
is called several times 

- https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/blob/cc6fb8ea272936550f65795791bf4e60d7955107/src/Mount.cpp#L212
- https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/blob/cc6fb8ea272936550f65795791bf4e60d7955107/src/Mount.cpp#L269
- https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/blob/cc6fb8ea272936550f65795791bf4e60d7955107/src/Mount.cpp#L269

The problematic call seems to be in line
- https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/blob/cc6fb8ea272936550f65795791bf4e60d7955107/src/b_setup.hpp#L312
while the corresponding driver is initialized starting at line 333:
- https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/blob/cc6fb8ea272936550f65795791bf4e60d7955107/src/b_setup.hpp#L333

So I deleted all calls to configHemisphere and only call it afterwards @ line 349.
https://github.com/OpenAstroTech/OpenAstroTracker-Firmware/blob/cc6fb8ea272936550f65795791bf4e60d7955107/src/b_setup.hpp#L349
Can someone please check if I missed something?

With these mods my eps32 build runs.
Thanks

ClutchplateDude, 2024-05-12:
Could you verify this with V1.13.3, please?