OpenAstroTech/OpenAstroTracker-Firmware #237: RA Steps getting re-set to absurd value

anjok, 2023-10-01:
Several times now I homed the OAT during the day prior to putting it outside and I had it going off the rails in RA. The RA steps were fine when I set it up.

Checking the RA steps when this happens gives me 13344.0 instead of the 417 it should be. The result is that the OAT moves a huge amount but thinks it only moved a bit (factor 32).

I haven't been able to reliable reproduce this though, I only guess that it's related to the time the mount it up.

I already switched the board, but it's still the same.

I'm using Firmware 1.13.5.beta

>> Command: XGR
>> Sending: :XGR#
<<Waiting for result: :XGR#>> newText=>Meade=13344.0
<< Received: 1 13344.0
# XGR -> (1, '13344.0')
>> Result: (1, '13344.0')
>> Command: XSR417
>> Sending: :XSR417#
<<Waiting for result: :XSR417#>> newText=>Meade=
<< Received: 3 
# XSR417 -> (3, '')

Blendan1, 2024-05-12:
I was able to replicate the bug, it seems to happen when loading the values from the EPROM and they are always off by the same factor.

When this happens Multiple times this might exceed the max value for the variable and stop the OAT from booting correctly.

anjok, 2024-05-12:
This should actually be fixed with the 1.13.3 firmware from today

ClutchplateDude, 2024-05-12:
Yes, this is fixed in V1.13.3 as Anjo mentioned.

closed 2024-05-12 by ClutchplateDude