OpenAstroTech/OpenAstroTracker-Firmware #246: #define NEW_STEPPER_LIB compiler FAILS - both Arduino IDE and Platform.IO

illigtr, 2024-06-01:
For some reason, I have NEVER been able to compile using the NEW_STEPPER_LIB definition. Always get a pack of errors. Why won't this work?

My Configuration:
 - MKS 1.21 with 3 Triminiac 2209 V3.0 with UART configuration. (works flawlessly). Requires special diagonal jumper.
 - Firmware version V1.13.5 (latest!)
 - OAT 3D printed with aluminium base and modified RA and DEC mounts with belt tensioners
 - 1.8 NEMA 17 steppers for RA and DEC, 
 - MPU4050 and GPS
 - RA and DEC Hall sensors for auto-homing
 - DF Robot LCD (16x2) Display with keypad
 - modified 28BYJ-48 stepper for DIY focuser
 - Pi4 8GB
     -  Bookworm OS
     -  running latest INDI drivers (including OpenAstroTeck Mount, and LibCamera)
     - Minicom and Python scripts to talk to the MKS for manual adjustments
     - 5Ghz WiFi
     - Pi HQ camera mounted on custom Newtonion scope (LibCamera)
 - Windows 11 PC running latest version of Kstars and EKOS and connected via 5 Ghz WiFi to Pi4

Here is my Configuration_local.hpp

```C
/////////////////////////////////////////////////////////////////////////////////////////////////////////
// This configuration file was generated by the OAT/OAM Configurator at https://config.openastrotech.com
// and is for firmware to be used on a OpenAstroTracker.
// Save this as Configuration_local.hpp in the folder where you placed the firmware code.
/////////////////////////////////////////////////////////////////////////////////////////////////////////
// Unique ConfigKey: TR:OAT,FWT:L,SL:N,HS:N,BD:M21,RST:N8,RDO:TU,RATP900:O90:A3000:V1800:S16:T256:,RTR:Y,RT:1,DS:N8,DDT:TU,DATP900:O90:A3000:V1800:S16:T256:,DT:1,DLIN90:D45:,STL:S,DY:NO,GP:N,LM:N,FC:Y,FS:BY,FD:TU,FAP150:O100:S1:H10:,FMSA600:V400:,APT:N,RAH:N
/////////////////////////////////////////////////////////////////////////////////////////////////////////

// Use the much higher performance stepper library.
//#define NEW_STEPPER_LIB

// We live in the Northern Hemisphere
#define NORTHERN_HEMISPHERE 1

// We are using the MKS GEN L V2.1 board
#if defined(BOARD) && BOARD != BOARD_AVR_MKS_GEN_L_V21
    #error Selected PIO environment does not match this configuration
#else
    #define BOARD BOARD_AVR_MKS_GEN_L_V21
#endif

////////////////////////////////
// RA Stepper configuration (OAT)
// See supported stepper values. Change according to the steppers you are using
// Using the NEMA 17, 1.8°/step stepper for RA
#define RA_STEPPER_TYPE STEPPER_TYPE_ENABLED
#define RA_STEPPER_SPR 200.0f

// Using the TMC2209-UART driver for RA stepper motor
#define RA_DRIVER_TYPE DRIVER_TYPE_TMC2209_UART

// Define some RA stepper motor settings
#define RA_MOTOR_CURRENT_RATING        900 // mA
#define RA_OPERATING_CURRENT_SETTING   90 // %
#define RA_STEPPER_ACCELERATION        3000
#define RA_STEPPER_SPEED               1800
#define RA_SLEW_MICROSTEPPING          16
#define RA_TRACKING_MICROSTEPPING      256

// Is it going the wrong way?
#define RA_INVERT_DIR  0

#ifdef NEW_STEPPER_LIB
  #define RA_SLEWING_ACCELERATION_DEG   2.0  // deg/s/s
  #define RA_SLEWING_SPEED_DEG          2.0  // deg/s
#endif

// Track immediately after boot
[... 335 lines cut ...]

//Pin to turn on dew heater MOSFET
#ifndef DEW_HEATER_1_PIN
    #define DEW_HEATER_1_PIN 10
#endif
#ifndef DEW_HEATER_2_PIN
    #define DEW_HEATER_2_PIN 7
#endif

//Serial port for external debugging
#if DEBUG_SEPARATE_SERIAL == 1
    #ifndef DEBUG_SERIAL_PORT
        #define DEBUG_SERIAL_PORT Serial2  //D16 (LCD_RS) - TXD2 and D17 (LCD_EN) - RXD2
        // #define DEBUG_SERIAL_PORT Serial3 //D14/15
	#endif
//#else
//    #ifndef DEBUG_SERIAL_PORT
//        #define DEBUG_SERIAL_PORT Serial
//    #endif
#endif

```

Beyond that, the only other mod is to adjust the values for the up/down/left/right/enter of the LCD keypad. 
Hijacked Serial2 for debug output and moved LCD pins to new pins. All good.

ClutchplateDude, 2024-06-02:
Since you didn't provide any errors themselves, there is no way for us to know what's happening. One thing that recently came to light is that the lowest RA_SLEW_MICROSTEPPING and DEC_SLEW_MICROSTEPPING need to be 32 (you have them at 16). Until you provide us actual errors to examine (the errors explain what the problem is), we can't help you.

closed 2024-06-02 by ClutchplateDude

illigtr, 2024-06-02:
I inadvertantly forgot the errors from the build. Mea culpa!

```
In file included from .pio\libdeps\mksgenlv21\InterruptStepper\include/Stepper.h:25:0,
                 from src\StepperConfiguration.hpp:13,
                 from src\Mount.hpp:15,
                 from src\HallSensorHoming.hpp:5,
                 from src\Mount.cpp:5:
.pio\libdeps\mksgenlv21\InterruptStepper\include/AccelerationRamp.h: In instantiation of 'class AccelerationRamp<256, 16000000, 314, 314>':
.pio\libdeps\mksgenlv21\InterruptStepper\include/Stepper.h:306:65:   required from 'static uint32_t Stepper<INTERRUPT, DRIVER, RAMP>::distanceToGo() [with INTERRUPT = IntervalInterrupt<(Timer)4>; DRIVER = Driver<Pin<60>, Pin<61> >; RAMP = AccelerationRamp<256, 16000000, 314, 314>; uint32_t = long unsigned int]'
src\InterruptAccelStepper.h:89:37:   required from 'uint32_t InterruptAccelStepper<STEPPER>::distanceToGo() [with STEPPER = Stepper<IntervalInterrupt<(Timer)4>, Driver<Pin<60>, Pin<61> >, AccelerationRamp<256, 16000000, 314, 314> >; uint32_t = long unsigned int]'
src\Mount.cpp:1444:66:   required from here
.pio\libdeps\mksgenlv21\InterruptStepper\include/AccelerationRamp.h:91:5: error: static assertion failed: Amount of steps per stair has to be at most 128
     static_assert(STEPS_PER_STAIR <= 128, "Amount of steps per stair has to be at most 128");
     ^~~~~~~~~~~~~
.pio\libdeps\mksgenlv21\InterruptStepper\include/AccelerationRamp.h:92:5: error: static assertion failed: Amount of steps per stair has to be power of 2
     static_assert(is_pow2(STEPS_PER_STAIR), "Amount of steps per stair has to be power of 2");
     ^~~~~~~~~~~~~
Compiling C:\Users\tony_\AppData\Local\Temp\tmphluryn50_patched_WInterrupts.c.o
*** [.pio\build\mksgenlv21\src\Mount.cpp.o] Error 1
================================================================================= [FAILED] Took 3.89 seconds =================================================================================

Environment    Status    Duration
-------------  --------  ------------
mksgenlv21     FAILED    00:00:03.885
```

I changed the minium slew microstepping as suggested to 32 and the program compiled flawlessly! Thank you. 
It should be noted that those settings for the RA and DEC motors comes from the OTA Firmware Configurator v2.18. Perhaps this program should be updated to set the default values to 32 instead of 16... 

[image]