OpenAstroTech/OpenAstroTracker-Firmware #255: Does not compile for some values of DEBUG_LEVEL

iThom96, 2024-12-19:
If I change the `DEBUG_LEVEL` from `DEBUG_NONE` to a value other than `DEBUG_STEPPERS` or `DEBUG_GUIDE`, the firmware does not build.

in `mount.cpp:1588` we have 
```
#if (DEBUG_LEVEL & (DEBUG_STEPPERS | DEBUG_GUIDE))
    const char *directionName = "-NE-S---W";
#endif
    LOG(DEBUG_STEPPERS | DEBUG_GUIDE, "[GUIDE]: guidePulse: > Guide Pulse %c for %dms", directionName[direction], duration);
```

So for values other than `DEBUG_STEPPERS` or `DEBUG_GUIDE`, `directionName` is not defined, but used in the next line, causing 

```
In file included from src/Mount.cpp:2:0:
src/Mount.cpp: In member function 'void Mount::guidePulse(byte, int)':
src/Mount.cpp:1591:89: error: 'directionName' was not declared in this scope
```

Elenedeath, 2024-12-19:
i can confirm and i can add when you use debug like
 `#define DEBUG_LEVEL (DEBUG_GENERAL)`
 
 doesnt compile but if you use
`#define DEBUG_LEVEL (DEBUG_GENERAL|DEBUG_STEPPERS)`

he work and compile.

i dont know if you are on the same board has me but i'm on MKS gen L V2.1

ClutchplateDude, 2024-12-20:
Fixed.

closed 2024-12-20 by ClutchplateDude